sharpie7/circuitjs1 #965: Voltmeter and Ammeter Current Flow Absent

Zachrack, 2024-10-25:
Hi, I set up a simple circuit with 2 voltmeters and I noticed at first that the voltmeters will function with no current flow. Then I tried a few things to try and trick the program into putting current flow through the voltmeters. I was unsuccessful. 

[image]

IRL, voltmeters and ammeters will have current flow when connected. Only after they are fully wired up should they measure anything. Maybe there can be a 1-terminal voltmeter that works like the current version and the 2-terminal voltmeter can work more like a resistor? The ammeter seems to work fine in normal circuits but will not short the power source and will not show current through a voltmeter.

Thanks!
[Circuit JS Meter Bugs.txt](https://github.com/user-attachments/files/17522280/Circuit.JS.Meter.Bugs.txt)

pfalstad, 2024-10-25:
Like many other components in this simulator, the voltmeters are ideal.  The voltmeter on the left has the lower terminal unconnected to anything.  Unconnected terminals are treated as ground, so that's why it's measuring 9 volts.

I could add an optional resistance to the voltmeter so it draws a small amount of current.

The ammeter can't short the power source because the power source and ammeter are both ideal so the current would be infinite.

Zachrack, 2024-10-25:
A very small resistance to the voltmeter would be excellent! I'm looking at using this software of yours as a teaching aid and having things closer to real life would be helpful. If the voltmeter has the resistor feature on I hope it would need to be grounded for the voltmeter to work.

Thanks again.

pfalstad, 2024-10-25:
I think you'd want the resistance to be really large, not really small, right?  Otherwise you'll get lots of current through the voltmeter which is not realistic.

Yes, having a resistor would fix the other issue too.  It would not work unless connected to something.

Zachrack, 2024-10-25:
You're right! (Low on sleep...) I've also noticed that the inductors and capacitors are ideal too, like you stated earlier. Maybe a 'non-ideal' menu option is in order? Haha. I'm happy if the voltmeter has an option to make it a bit more realistic. I really do appreciate it.