sharpie7/circuitjs1 #967: Bug-analyzeCircuit() executes when simulation is disabled

Taste62, 2024-11-01:
Circuitjs executes analyzeCircuit() after every change in the layout, which makes it impossible to work on larger projects, even, if the simulation times afterwards are ok.

pfalstad, 2024-11-01:
could you send me an example of a large file that analyze circuit runs slowly on?

Taste62, 2024-11-02:
I did some more testing and there seems to be more depth to this than I expected. anaylzeCircuit() takes 7800ms in [LaggyCircuit1.circuitjs.txt](https://github.com/user-attachments/files/17606662/LaggyCircuit1.circuitjs.txt), but when I load the subcircuit file it drops to 50ms. [LaggyCircuit2.circuitjs.txt](https://github.com/user-attachments/files/17606660/LaggyCircuit2.circuitjs.txt) is just a long wire with a resistor and two labels on each end. In this file analyzeCircuit() takes around 16000ms, but when I put it into a subcircuit, it drops down to 0.2ms (see [LaggyCircuit(2).circuitjs.txt](https://github.com/user-attachments/files/17606661/LaggyCircuit.2.circuitjs.txt)).

pfalstad, 2024-11-02:
Yes, I tried a test circuit with a large number of wires chained together, and that was really slow because of the wires, but I wasn't sure how realistic that situation was.  When you put it in a subcircuit the wires are removed so it's fast.

Taste62, 2024-11-02:
How about LaggyCircuit1? Even, if it's expected to be so slow, it would help a lot, if it only analyzed the circuit when the simulation is running, so the circuit can be modified without waiting a few seconds every time something is moved.

pfalstad, 2024-11-02:
Yeah, it doesn't need to analyze the circuit when it's not running.  So that would be an easy fix, I assume.

LaggyCircuit1 is slow because of findUnconnectedNodes().  The main loop is slow and has to be run ~20 times.  I'm not sure we need to start over from scratch every time we find an unconnected node, so I may be able to speed that up.  Also, finding connections between two nodes looks really slow for subcircuit elements.  Maybe that could be speeded up or at least cache the results.