sharpie7/circuitjs1 #968: RMS value is dependent on time division

by-justin, 2024-11-03:
The Vrms value changes when I zoom in/out of the scope or change the horizontal scale in scope properties. This sine voltage source displays 2.977 Vrms when time division is set to 1ms/div and 3.386Vrms when set to 500us/div.

[image]

Moreover, Vrms value will not display on high frequency circuits. Adjusting timestamp size changes rms value as well.

[circuit-20241103-0940.circuitjs.txt](https://github.com/user-attachments/files/17611234/circuit-20241103-0940.circuitjs.txt)

pfalstad, 2024-11-03:
Yes, that's true.  This isn't really fixable.  We calculate Vrms from the data we have available.  If the scope is zoomed in, there's more data for each cycle so the value will be more accurate.  Also if the timestamp size is smaller, the value will be more accurate for the same reason.

by-justin, 2024-11-04:
Thanks for the clarification!

closed 2024-11-04 by by-justin