sharpie7/circuitjs1 #971: Bug - Inductor reset is silent: energy conservation violated without warning

calimeroteknik, 2024-11-21:
I noticed an energy conservation issue in this trivial circuit:
[image]
We can observe that when the switch is closed then opened, it dissipates no power; so where has the energy gone?

Thinking what ***should*** happen: if all components are ideal, we would get infinite voltage across the inductor.
Which means "overvoltage" in this simulator, and would typically halt the simulation.

Investigating the reason for the present behaviour of the simulator, a clear trend emerges:
https://github.com/sharpie7/circuitjs1/issues/799 :
>It seems that the simulator can't find a current path for the inductor, so […] it resets the inductor.

https://github.com/sharpie7/circuitjs1/issues/675 :
>The simulator doesn't think one of the inductors has a current path, so it zeroes out the current. […]
>There is probably a better way to fix this.

I suggest that a better way to fix this is to interrupt the simulation. Teleporting it silently to a different state isn't a good idea!

It could be as trivial as a pop-up that says it happened, so that at least the physics law violation isn't happening silently.
I think the patch is something trivial like:
```patch
+alert("Resetting inductors because no current path was found. Energy conservation will break.")`
```

The helpfully informed user can then reflect upon his design choices, or insert a MΩ resistor, pF capacitor or such, to simulate the real circuit better in such a circumstance if desired!

esaruoho, 2026-03-03:
This is a known limitation of how the simulator handles inductors with no current path. When the circuit topology changes (e.g., a switch opens), the inductor current is reset to prevent singular matrix errors, but this silently violates energy conservation. The v3-dev branch has improvements to convergence handling and capacitor series resistance that help in some cases, but the fundamental inductor reset behavior remains.

calimeroteknik, 2026-03-03:
Any feedback on my suggested improvement?
Bottomline, it's entirely acceptable that the simulator would be unable to handle that case and I never meant to request any improvement to the simulator itself, but I believe the simulator should tell the user (*and then* still reset the inductor current), instead of resetting it ***silently*** mid-simulation.