sharpie7/circuitjs1 #972: Application stops working when opening Unijunction Oscillator

vishnu-v-vardhan, 2024-12-01:
The website just stops responding when trying to open a Unijunction Oscillator from **Circuits > Misc** section.

The **GET** request seems to be fetching the txt file for the oscillator circuit but upon execution, the entire site hangs and stops responding. Tried pasting the following code into a new circuit but got the same output.

$ 65 0.0000049999999999999996 0.23988752939670982 62 5 50 5e-11
r 496 176 496 48 0 470
r 352 48 352 176 0 100000
c 352 176 352 304 0 1e-8 5.060465273168598 0.001
r 496 208 496 384 0 47
g 496 384 496 400 0 0
g 352 368 352 400 0 0
R 352 48 272 48 0 0 40 10 0 0 0.5
w 352 48 496 48 0
r 352 304 352 368 0 1
417 416 176 496 176 1
r 352 176 416 176 0 10
o 2 2 0 4099 10 0.05 0 2 2 3
o 10 2 0 4097 0.625 0.05 1 2 10 3

pfalstad, 2024-12-01:
oops..  will fix shortly

pfalstad, 2024-12-01:
fixed in 3.0.2

vishnu-v-vardhan, 2024-12-02:
The issue still persists for the same circuit.

[image]

pfalstad, 2024-12-02:
try shift-reload.
do File->About to make sure you are running 3.0.2.

vishnu-v-vardhan, 2024-12-10:
Yep! It's working! Thank you for the quick fix and for such a great tool!

closed 2024-12-10 by vishnu-v-vardhan