sharpie7/circuitjs1 #975: Feature - in v 3.0.2 needs a way to choose directory for saving project (on windows) and overwriting existing file

vgreff, 2024-12-02:
The latest version now  just saves in the windows download directory without asking. It no longer opens a dialog box to change the directory to save the file and it also keeps creating new versions of file for every save (same name and adds (##) suffix. 
There is no way to replace/overwrite the previous file.
A menu option to choose the working directory would be nice and an option to allow overwriting an existing file

pfalstad, 2024-12-02:
Hmm, that's odd because I didn't change anything related to this.  I just changed the filename format.  I'm also not sure if I have any control over whether the browser decides to display a dialog box.

This is the online version, right? What browser is it, Chrome?

pfalstad, 2024-12-02:
I tried firefox on windows and it acts the same with the old version.

You can try 2.9.0 here: https://www.falstad.com/circuit/circuitjs96.html

It's the browser that's asking what directory to save in.  The web app has no control over that.  It can't check to see if there's a file there already either.  For security reasons, the web app can't look at your files.  I think there might be a new extension which allows that but I don't think firefox supports it anyway.

Maybe there is a browser setting that controls that, which got changed recently?

vgreff, 2024-12-02:
Problem fixed, I can confirm now and  that the behavior in 3.0.3 is the same to 2.9.1. Dialog boxes are all back.

Polarized caps fixed worked too BTW as you said in other mail

 

I had to manually clear 7.2MB of data that associated with Falstad.com in the firefox browser on my windows 10 machine and then the 3.0.3 version showed up there as well. Without manually clearing it did not auto upgrade to 3.0.3, not a big deal but I wanted to share that data point.

 

Vincent B. Greff

pfalstad, 2024-12-02:
Yeah I noticed there's something going on with the caching, I need to figure that out