sharpie7/circuitjs1 #976: Feature - copy and paste a selected section of a circuit from 2 instances of the simulator nto another (almost possible, but not quite)

vgreff, 2024-12-02:
It would nice to be able to  export just a selected portion of the circuit as text (since copy paste does not work between 2 tabs in the browser or 2 instances of the browser).
This would allow copying  a portion of a schematic from one circuit into another circuit in another tab.

Unfortunately currently, the import from text also clears out the existing circuit in that instance. It would nice to be able to import into the loaded circuit without clearing clearing it first.

pfalstad, 2024-12-02:
copy/paste works between 2 tabs for me.  I haven't tried firefox on windows but it works on mac firefox.

vgreff, 2024-12-02:
My bad, it works for me now as well in 3.0.3. 

I tried it this past weekend I could not get it to work then.

I’m sorry I should have re-tried it with v3.0.3 after clearing the cache before posting the feature request.

Maybe that had a weird interaction.

Any thank you for the quick response.

 

One other feature idea I had, was be able to have some user configurable icon in your new toolbar for non-common components used in the current design. For example I was using Schmidt trigger and real amp ops and they don’t have a shortcut. I know I can assign a shortcut for it but would be nice if I could click on instance in my current schematic and assign it to a user defined slot  and use the slot as a shortcut too (may use numbers 0-9) and they would on the toolbar too under a dropdown like the logic gates for easy selection. 

 

Vincent B. Greff

closed 2026-05-04 by pfalstad