sharpie7/circuitjs1 #979: BUG? INCORRECT AC RMS VOLTAGE

terrysteven, 2024-12-30:
AC RMS VOLTAGE FLUCTUATES BASED ON HORIZONTAL SCALE SETTING, RESULTING IN INCORRECT VALUE.

EXAMPLE. INCORRECT VRMS ON LEFT SCOPE. CORRECT ON RIGHT SCOPE. THIS IS VERY FRUSTRATING AS THE HORIZONTAL SCALE HAS TO BE CONSTANTLY ADJUSTED TO ZERO IN ON THE CORRECT RMS VOLTAGE.
https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKADdxCrMVOqwuUcMjjCq46AhYAnPnOwJeA8SMgsA5iAVLB28MV7iNcnvMgAWIWoAeIBimy8ueEOZeFmr8LwDqACQBNAB0AZ3NKUIBbADVpSJCAflCACj8AeQAlAEkALTSAOQAVAEEAGVCAZQBhMoBRULBIeIB6ABMASzYAShYAeyFCS25IHBckGAhuIWw+oUmQNFGFyShoCAhDLRYgA

terrysteven, 2024-12-31:
I can type faster in caps. I am not yelling.  Anyway here's some lower case letters for you.

"Show Extended Info"   gives   rms values for  Voltage sources only, not   for other components like resistors (way more important).  In the example the horizontal scale needs to be moved all the way to 100us/div to get the correct rms voltage. I know how to "slide" into the correct  rms value. What good is having a number that is "closer" but still way off? Have you ever worked with AC current? Have you ever tried to get the correct  AC voltage drops across resistors? This is very basic stuff.  I have to move the horizontal slider every single time I want the correct rms measurement.  In my example the 707mv reading from the right scope  is correct and  450mV reading from the left scope just wrong. Why bother putting an incorrect value in the first place.   I guarantee a lot of people using this SIM probably don't understand they are seeing   incorrect rms readings.

Rms calculations should not depend on time per division. The sim just needs to take say 12 equidistant points from only 1 cycle and then put those in the rms formula vrms = sqrt((v1^2+v2^2+v3^2+............v12^2)/12) and show the value, updating (getting a new 12 point sample) say every 1-2 seconds. Yes even this small sample size from a single cycle would yield very accurate (good enough for a SIM) results.  This would be a   fast easy calculation for the sim to make. No stress, no confusion.

pfalstad, 2024-12-31:
The RMS voltage is calculated from the available data and higher resolution data gives a more accurate value.  The resolution of the data depends on the horizontal scale.

We don't store all the data points for every time step.  Instead we average it down according to the scope scale, to save memory.  We should probably just store all the data points.  The calculated info like RMS voltage, frequency, spectrum, etc. would be more accurate and the simulation would probably run faster because less work is done at every time step.

pfalstad, 2024-12-31:
Calculating the RMS voltage is easy if you happen to know the frequency, but we don't know that in general so we only do it for AC sources.

terrysteven, 2024-12-31:
Hi
Thanks for the reply. Believe me I really like this software but it would be nice if you could mouse over  a resistor when AC is flowing and get the True rms in the info box, instead of the rapidly changing numbers.  DC info is very good,  you instantly know the current and voltage and power with a simple mouse over. For audio I need to measure AC  rms all the time. Would be nice if the info box could show pulsating DC, Max/Min Amp and true Arms and Vrms. 

For voltage sources I assume software multiplies Amplitude x 0.707. 

There is no data base at all with this software? I am not a software expert but I understand a "mini" database can be put in  a browser while the software runs.  Is that true? How much work is that?

Yeah I can see when the software runs  the voltage values (in the info box) are being calculated in real time and displayed.  Those values would fit perfectly into the true rms equation.

pfalstad, 2024-12-31:
What do you mean about the mini database?  What do you want to put in it?

Not sure if we want to store all past per-timestep data for all elements.  For huge circuits that would be bad.  But maybe up to a certain limit.  Or, when mousing over an element, it would start storing the data so it could display the RMS voltage if you hold for a few seconds.

terrysteven, 2024-12-31:
By mini database I mean something like: IndexedDB API is client-side storage for storing structured data. IndexedDB is a transactional database system like an SQL-based Relational Database Management System, but data is stored in JavaScript objects.

I thought sample voltage values eg V1, V2,......Vn for a single cycle would need to be stored in order to calc rms ? i am just guessing,  I am not sure exactly what or how data gets stored  on a browser.

The mouse over sounds like a good idea.  I think there is a lot of potential there.

pfalstad, 2024-12-31:
yes the voltage samples would have to be stored.  that's not hard.  a relational database would be overkill for that.