sharpie7/circuitjs1 #985: Purpose of Invert Inputs

ArmoredRodent, 2025-01-21:
Not an issue, but what is the purpose of Invert Inputs on NAND, AND, NOR and OR gates?  I've been doing this a while and I am confused of the purpose.  In principle, I could see inverting one input to eliminate an inverter, but not really sure of all.  I've disabled it on my local version and replaced it with DeMorgan's Symbol option.

Attached text file shows a NAND with and without Invert Inputs selected.  If you mouse over Invert Inputs NAND, it says it is a NAND, but logically with inverted inputs, it is an OR. A tad confusing and I'm not sure how it can be used.

Is this a partial implementation of DeMorgan's gates?

I have the logic worked out for [DeMorgan's gates](https://wulflemm.com/war/circuitjs.html?startCircuit=deMorgan1.txt) (version 3.1.1).  Interested?

```
$ 1 0.000005 10.20027730826997 50 5 50 5e-11
151 272 176 368 176 0 2 0 5
151 272 272 368 272 4 2 5 5
L 144 208 96 208 0 1 false 5 0
L 144 240 96 240 0 1 false 5 0
M 368 176 432 176 0 2.5
M 368 272 432 272 0 2.5
207 144 208 192 208 4 A
207 144 240 192 240 4 B
207 272 160 224 160 4 A
207 272 192 224 192 4 B
207 272 288 224 288 4 B
207 272 256 224 256 4 A
```

pfalstad, 2025-01-21:
There are some schematics which show a NAND with inverted inputs in place of an OR gate (for example) so I wanted to make it possible to represent those in the simulator.  Rather than have an alternate symbol for the OR gate, I decided to make inverted inputs an option.  It seemed easier to explain, maybe.

ArmoredRodent, 2025-01-22:
I sort of figured that.  I did a quick look before I posted.  I'll grep the txt files and do a tad more exploring.

ArmoredRodent, 2025-01-29:
I grepped your text files and Invert Inputs are not used. So no impact correcting this by transitioning to DeMorgans. 

Invert Inputs changes the underlying logic without changing the image.  A NAND with inverted inputs becomes an OR, but looks like a negative-input NAND.  For me, this makes it very doubtful that anyone uses Invert Inputs. 

A DeMorgans NAND becomes a negative-input OR.  The symbol changes, but it is still a NAND.  Anyone understanding DeMorgans gates looking at an inverted input NAND would think the underlying logic is an OR, which adds a layer of confusion. 

On rare occurrences, I have seen MSI chips, where one input is inverted, but never two. I think implementing that is more trouble than it is worth. At the gate level, anything is possible. 

DeMorgans gates are a visualization of DeMorgans Theorem. That has very little to do with how they are used. 

The internet calls it bubble pushing or matching. That name sort of misses the point for me.  Required inputs to match output.  Not very catchy as a name.

You can see it in [74ALS138 - 3 to 8 decoder](https://wulflemm.com/war/circuitjs.html?startCircuit=deMorgan7.txt). 

[Diodes 74HC138](https://www.diodes.com/assets/Datasheets/74HC138.pdf)

[TI 74LS138](https://www.ti.com/lit/ds/symlink/sn54ls138-sp.pdf)

Standard gates are active-high inputs, while DeMorgans gates are active-low inputs. Both are used to show the default inputs/outputs in a combinational circuit to activate an input. Sort of built in documentation.  Even negation of a signal can be easily identified.

The use of both allows for troubleshooting without understanding the logic.  An active-low chip select input requires specific active-low or active-high inputs.

DeMorgans gates is one of the first things I look for in a schematic capture package (Kicad for example), even though it has little use today.  So it had to be one of the first things I implemented when I had my own local version of your simulator.

pfalstad, 2025-01-29:
Well, someone might be using it somewhere.  I wouldn't introduce an incompatibility.

I looked for "DeMorgan's gates" and couldn't find it anywhere.  So I'm not sure if that's going to be meaningful to people.  The meaning of "invert inputs" is obvious.

"NAND with inverted inputs becomes an OR, but looks like a negative-input NAND."  Yes...  And an OR with DeMorgan's symbol looks like a negative-input NAND.  I don't understand the difference.

ArmoredRodent, 2025-01-29:
You'd probably have more success searching with bubble pushing.

I guess my point is you are changing the underlying nature of the gate without a backup by theory. 

> $ 1 0.000005 10.20027730826997 50 5 50 5e-11
151 272 176 368 176 0 2 0 5
151 272 272 368 272 4 2 5 5
L 144 208 96 208 0 1 false 5 0
L 144 240 96 240 0 1 false 5 0
M 368 176 432 176 0 2.5
M 368 272 432 272 0 2.5
207 144 208 192 208 4 A
207 144 240 192 240 4 B
207 272 160 224 160 4 A
207 272 192 224 192 4 B
207 272 288 224 288 4 B
207 272 256 224 256 4 A
x 366 157 434 160 4 24 NAND
x 361 252 643 255 4 24 NAND\swith\sInverted\sInputs
152 272 368 368 368 0 2 5 5
M 368 368 432 368 0 2.5
207 272 352 224 352 4 A
207 272 384 224 384 4 B
x 364 354 400 357 4 24 OR
I 208 64 272 64 0 0.5 5
I 208 96 272 96 0 0.5 5
152 272 80 368 80 0 2 0 5
M 368 80 432 80 0 2.5
207 208 96 160 96 4 B
207 208 64 160 64 4 A
x 363 60 549 63 4 24 DeMorgan\sNAND

pfalstad, 2025-01-29:
Nobody asked for this feature so I was looking for something easy to implement.  Changing an AND to an OR required more rearranging of code.

You make a good point about the mouse-over text being confusing.  That should probably be changed.

ArmoredRodent, 2025-01-29:
hehehe Somehow I have made it worse.

pfalstad, 2026-09-05:
I merged your changes which resolved this issue, I think.

closed 2026-09-05 by pfalstad

ArmoredRodent, 2026-09-05:
I'll check all my changes over the next couple of days and modify my circuits for the new version.  Thanks.