sharpie7/circuitjs1 #988: Absurdly huge voltages

MaSo-CZ, 2025-02-01:
A simple circuit with operational amplifiers and TL431 ends soon after starting with a red stop sign and shows absurdly huge voltage values. The power supply is only 12V and the battery is 4V. I put the -12V source only for troubleshooting. Resistors in the operational amplifier power supply are also used for troubleshooting. Previously, I got absurd giga amps there.

Help please :)

[image]

[circuitjs-20250201-175853.txt](https://github.com/user-attachments/files/18629582/circuitjs-20250201-175853.txt)

pfalstad, 2025-02-01:
Hi,

It seems like the complexity of the components in this circuit—especially the real op-amps and the TL431—may be pushing beyond the intended scope of this simulator. The simulator is primarily designed for teaching purposes, where most components are modeled as idealized versions rather than fully accurate real-world equivalents.  If you need precise modeling of these components, I’d recommend trying a simulator like LTspice.  The simulator is definitely doing something wrong here but I'm not sure that debugging it would be productive.

Otherwise, I would take out the TL431 because you don't need one.  All the voltage sources in this simulator are perfectly regulated.  Then, replace the real op-amps with ideal ones, and see if it works.

mpictor, 2025-09-18:
@MaSo-CZ as mentioned in the linked issue, it may help to add series resistance to each opamp output.