sharpie7/circuitjs1 #989: BJT keeps conducting, even when $\ V_{BE} < 0.7V \$

antoniu200, 2025-02-07:
Hello,

I am trying to create some simple logic gate circuits, but it seems I am encountering an issue with NPN BJTs.
Even though $\ V_{BE} < 0.7V \$, the component's $\ V_{CE}\$ refuses to climb (i.e. BJT is not closing).

[image]

I am a beginner in electronics, so please go easy on me.
However, I think this is abnormal, as an NPN BJTs only opens when $\ V_{BE} > ~0.7V \$.

Could somebody please take a look at this and confirm?

Code below:
[test gate.txt](https://github.com/user-attachments/files/18707252/test.gate.txt)

pfalstad, 2025-02-07:
The transistors actually aren't conducting (it's in cutoff mode).  But all their terminals are at 5V, so it doesn't matter if it's conducting or not.  What kind of gate are you trying to build?

antoniu200, 2025-02-07:
I was trying to derive an AND gate from an OR gate, but this was a massive fail. After maybe an hour of analysing some tutorials online, I managed to build a circuit that worked.

Now, after trying this same thing again, I realized the mistakes I did, why and corrected them.
What can you do... This is my learning curve...

Oh, well. I suppose it not an issue with the simulator, which is good to know.

Thanks for your input anyway!

closed 2025-02-07 by antoniu200