sharpie7/circuitjs1 #990: Oscilloscope connection

Nes-si, 2025-02-08:
Is it possible to connect oscilloscope to two selected points on circuit? For example, I've tried to connect osc to transformer winding and I couldn't.

pfalstad, 2025-02-09:
You need to connect the two points with a voltmeter / scope probe.  Then you can view the voltmeter in a scope.

pfalstad, 2025-02-16:
You can use an ammeter to get the current.  You can view that separately or combine the plots into one scope.

I should add a way to view the voltage/current of one of the windings in the scope.

Adding a way to connect a scope between two arbitrary points sounds complicated.  The voltage is easy to get of course, but not the current.

Nes-si, 2025-02-16:
@pfalstad understand.