sharpie7/circuitjs1 #991: Bug: wrong lamp resistance

Nes-si, 2025-02-16:
I add a lamp with default props (100 W 120 V): it means the lamp has a resistance of 144 Ohm when hot. So its resistance shown in info panel is seems like correct (7,2 Ohm when cold):
[image]

Next I connect it to AC source. Then I stop the simulation and reset. Now the info panel says the lamp has 100 Ohm resistance and T = 300 K (cold):
[image]

That's wrong, it should be 7,2 Ohm. After that I select Blank circuit, add a lamp, and the info panels tells me that the lamp has 0 Ohm resistance:
[image]

pfalstad, 2025-02-16:
Hmm, yes, when we reset the lamp or create a new one, it doesn't update the resistance right away.  It gets updated after the simulation starts running.

ArmoredRodent, 2025-02-25:
Just like a real bulb, it takes a while to come up to temperature and for the resistance to decrease and current to go up. 

The variable warmup time (400 ms) can speed up this transition.

pfalstad, 2025-02-25:
I think he understands that the resistance changes.  The issue is that the resistance isn't updated sometimes when it should be.  It doesn't affect the simulation, but it does show an incorrect value in the info panel.