sharpie7/circuitjs1 #993: Bug: updating voltmeter indications

Nes-si, 2025-02-17:
This simple circuit was created to watch capacitor discharge process. But when you switch off charged capacitor from AC power source (the capacitor starts to discharge), there will be no voltmeter indication updating:
[image]

pfalstad, 2025-02-17:
Hi, could you do File->Export As Link or Export As Text and post the circuit

Nes-si, 2025-02-17:
@pfalstad 
```
$ 1 0.000005 10.20027730826997 50 5 43 5e-11
v 192 272 192 368 0 1 40 5 0 0 0.5
s 192 272 304 272 0 1 false
c 304 272 304 368 4 0.00001 5.640504507908872e-7 0.001 0.1
w 192 368 304 368 0
p 384 272 384 368 3 1 0 10000000
r 496 272 496 368 0 1000
w 304 368 384 368 0
w 384 368 496 368 0
w 496 272 384 272 0
w 384 272 304 272 0
o 4 64 0 4098 5 0.1 0 1
```

pfalstad, 2025-02-17:
When measuring RMS voltage, the voltmeter is slow to react to changes because it has to average over several cycles.  It reacts pretty quickly to changes in voltage as long as the frequency doesn't change.  But if you turn off the AC source completely, it takes a while.

Increase the simulation speed to the maximum, and then you won't have to wait very long.

Nes-si, 2025-02-17:
@pfalstad yep, it changed indications after about 250 ms.