sharpie7/circuitjs1 #994: Error message: "Can't load circuit list!"

Nes-si, 2025-02-17:
When I export any circuit (even blank) to link and open it, I see a message: "Can't load circuit list!"
(I'm using Windows version.)

pfalstad, 2025-02-17:
So you get that message when you click on this?

https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcQUUAWcb37ITzgMhKMjgsgA

Can you click on this?

https://www.falstad.com/circuit/circuitjs100/setuplist.txt

Nes-si, 2025-02-17:
> https://www.falstad.com/circuit/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcQUUAWcb37ITzgMhKMjgsgA

There is no such message.

My link looks like this: `file:///C:/programs/circuitjs1/resources/app/war/circuitjs.html?ctz=CQAgjCAMB0l3BWcMBMcUHYMGZIA4UA2ATmIxAUgoqoQFMBaMMAKACcQ09wNDP9OeblTDwWQA`

> Can you click on this? https://www.falstad.com/circuit/circuitjs100/setuplist.txt

Yes, and?..

pfalstad, 2025-02-17:
ok I see..  "Export As Link" doesn't work in the offline version.