sharpie7/circuitjs1 #995: Watt-meter does not work as a watt-meter

ArmoredRodent, 2025-03-11:
A watt-meter has two meters: ammeter in series and a voltmeter in parallel.

This allows it to be connected to measure power in dc, single-phase ac and three-phase delta and wye ac.  This is explained very well and concisely in the linked Tektronix Application Note.

[Tektronix - The Fundamentals of Three-Phase Power Measurements - Application Note](https://download.tek.com/document/55W_28943_0_HR_Letter.pdf)

There are no examples on how to connect the watt-meter.  

Through a process of trial and error I have determined the following.  Please correct me if I am wrong.

The watt-meter has four terminals.  The upper two are connected in series with the load.  This would correspond to the watt-meter ammeter setting, but because the simulation has access to voltage, power can be calculated, which means that these are the only two terminals that must be connected.

The lower two confused me (and others).  

[reddit: Falstad circuitjs: What are the four nodes in wattmeter for?](https://www.reddit.com/r/AskElectronics/comments/1cfq1jh/falstad_circuitjs_what_are_the_four_nodes_in/)

I tried any number of possibilities and failed to figure out how to connect the lower two.  I believe I eventually found the solution via the web.  The lower two are connected with in series with the lower part of the load to a common or ground.  This makes the voltage across these two terminals 0V.  So very unclear how this works as a voltmeter in parallel with the load.  Essentially, the watt-meter relies on upper two terminals to measure power.  Lower two have no point.  Works well for dc and single-phase circuits, but it is not a watt-meter.

The two connections are shown in the [attached circuit](https://wulflemm.com/war/circuitjs.html?startCircuit=Changes_Wattmeter.txt) measuring three-phase wye-connected power.  Again, please correct me if I am wrong.

Since there is no true parallel voltage measurement, this severally limits the watt-meter's applications for three-phase circuits.  No delta loads.  No two watt-meter method.  In delta, there is no common on load side.  

I'd like to implement some of the Tektronix circuits.  

You can do it or I believe I can make a first pass at fixing the watt-meter, which you can improve upon.  But I don't want to step on your toes, if you don't want it. I have implemented RMS Power [#129 - Wattmeter Average Reading](https://github.com/pfalstad/circuitjs1/issues/129).  

Proposed changes:

If the bottom terminals are unconnected or voltage across the terminals is 0V, watt-meter defaults to existing behaviour and image.  Keeps existing usage happy.

If there is a voltage across lower terminals, image changes and correct watt-meter operation.  Tektronik's uses Hi and Lo, with the voltmeter after the ammeter to counter ammeter losses.  We'd use + on source side for ammeter and + on load side (opposite side) for voltmeter.  This minimizes simulator wires.

Goal: Implement some or most of Tektronix circuits.

Questions, thoughts, concerns...

pfalstad, 2025-03-13:
True, the wattmeter is not good and does not resemble a real wattmeter at all.  Sorry about that.

I'd be tempted to make a new wattmeter class from scratch and keep the old one only for backward compatibility.

If you are willing to, I'd appreciate it.

ArmoredRodent, 2025-03-13:
Not a problem that it is what it is.  Without an applicstion in mind it can be difficult to see future problems. Sort of you cannot see the forest because of the trees. 

New watt-meter class makes sense because kludging it would be a pain. 

I'll do a first pass and you can tweak it as required.

esaruoho, 2026-03-03:
pfalstad/circuitjs1 PR #236 (merged) adds wattmeter averaging to smooth the instantaneous power display, which helps for AC circuits. However, the deeper redesign discussed here (separate voltage/current sensing terminals) would require a new element class.