sharpie7/circuitjs1 #997: Very high power LED

JacobIsrael42, 2025-03-25:
Lately, at least, standard LEDs seem to draw a lot of power, even at 4Volts.

Thanks,
JACOBISRAEL040508@GMAIL.COM
Github/jacobisrael42

[Power LED.txt](https://github.com/user-attachments/files/19457377/Power.LED.txt)

pfalstad, 2025-03-25:
I think if you try that with a real LED you'll get the same result, except it will fry the LED.  You should put a resistor in series with it.

JacobIsrael42, 2025-03-26:
Hi Paul. I am very surprised that you said that. I have real LEDs here...using 3 Volts. They don't fry or heat up. If you like, I can put a 1 Ohm resistor in series to get the current. But, I believe that LEDs saturate at roughly 20-40 mA (within rated voltage).

tubesandmore.com/products/led-5mm-water-clear-high-brightness-package-5

pfalstad, 2025-03-26:
The default LED's in this simulator have a much lower voltage drop.  Here is one that might work better for you:

https://tinyurl.com/22qbsy7k

But, LED's are diodes so the current increases exponentially as the voltage increases.

JacobIsrael42, 2025-03-26:
Hi Paul,
I looked at the datasheets more carefully. I see that a standard LED will already be in thermal runaway at 3V. So, sorry to bother you.

I will try this LED.

closed 2025-04-17 by JacobIsrael42