sharpie7/circuitjs1 #1002: Option to CMOS 555

rmlazzari, 2025-05-11:
Hello, may I suggest adding the component library to the CMOS/FET version of the 555?

I apologize for giving my opinion without technical knowledge, but perhaps this option could be made in a small "Options" or "Properties" box for the component, like the one for editing diode properties, for example.

It would be very useful for those who need to simulate frequencies above those that the 555 BJT can generate.

Thank you.
Renato M. Lazzari

pfalstad, 2025-05-11:
The 555 in the simulator is idealized and does not use a BJT.  It should be able to generate any frequency you want, if the time step is small enough.  Are you having problems with it?

rmlazzari, 2025-05-11:
Thank you for the answer, dear pfalstad. And yes. I'm just an amateur - without English as the best idiom, sorry for that - but I'm trying to build an oscillator with frequency an duty cycle variables ( here: https://tinyurl.com/2xwtnacl ) in a range, say, between 100KHz and 500KHz / 10% to 100% (or something like that). But I think I've reached limits with something like 12KHz...

I've made some changes in the circuit as it is in the link, for example, R between pin 7 and VCC changed to 500Ω, but I didn't found how to record that change. No problem with that, I can write down in a paper when it works fine.

The main request is to get frequencies above the 12KHz I've told you...

rmlazzari, 2025-05-11:
OFF TOPIC: I'm grateful for your superfast answer and I'd really like to stay here to Q & A but I have to leave now (it's Mother's Day in my country). But sure I'll be back to see your answers and may be some tip that can help me with  the oscillator I need. And please feel free to ask me something you think could be useful for you.

Thank you a lot, pfalstad!
Best regards!

pfalstad, 2025-05-11:
I had to leave for Mother's Day too.

The simulator uses a timestep to run the simulation.  It measures the time difference between simulation steps.  The default is 5 microseconds which is too large to simulate a 500kHz oscillator.   To simulate an oscillator like that you probably want 20 or so steps per cycle, so a better timestep is 100 ns.  Go to Options->Other Options and set the time step size to 100n.  A smaller timestep will be more accurate but will also be slower.

https://tinyurl.com/2497jhu9

rmlazzari, 2025-05-12:
Sorry, but since I wasn't familiar with this powerful program, I missed the "Options" menu. If it's not too much trouble, could you point me to some good tutorials and manuals, so that I don't bother the forum with beginner questions from now on?

And thank you very much, pfalstad, it's working perfectly now.