sharpie7/circuitjs1 #1003: Floating-point value in diode model ignored (when contains exponent?)

geometrian, 2025-05-16:
I have a diode with parameters:
• I<sub>S</sub>&nbsp;&nbsp;= 4.416 409 060 438 224e-8 [A]
• R<sub>S</sub>&nbsp;= 0.117 918 182 847 822 07 [Ω]
• n&nbsp;&nbsp;&nbsp;= 6.446 193 498 698 821

However, when I enter these values into the advanced diode model, I<sub>S</sub> doesn't 'take'. It silently remains at what it was before! I assume this is because it fails to parse the "e-8" in string "4.416409060438224e-8".

The user notices that the model updated (as the other two parameters changed), but would have to go back into editing the model again to discover that it had only *partially* updated! The certainty of an error coupled with the unlikelihood of it being discovered makes this fairly serious in my view.

pfalstad, 2025-05-16:
Yes, it can't parse the scientific notation.  It should complain at least rather than silently ignoring it.

esaruoho, 2026-03-03:
PR submitted to fix this: https://github.com/pfalstad/circuitjs1/pull/282

The root cause is in EditDialog.parseUnits(). GWT's NumberFormat.parse() with the format "####.##########" cannot parse scientific notation like 4.416e-8. The exception is silently caught in apply(), so the value stays unchanged with no feedback to the user.

The fix adds a regex check before the unit-suffix parsing: if the input matches scientific notation (e.g. 4.416e-8, 1.2E+3, 5e9), it uses Double.parseDouble() which handles it natively. This applies to all edit dialogs, not just the diode model dialog.